Sevenstax/FreeV2G #319: Sequence error getting in xuv400

nikunjp26, 2024-05-30:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

**Dear Team**

We are trying to charge the XUV 3XO car of Mahindra but it gives the sequence error every time so can you please look into the and tell us quickly.

[mahindra_xuv_3XO_sequecne_error.zip](https://github.com/Sevenstax/FreeV2G/files/15495395/mahindra_xuv_3XO_sequecne_error.zip)

**Best Regards,
Nikunj Patel**

lho-stx, 2024-05-30:
Hey @nikunjp26,

per ISO15118-2 the allowed next request after a valid ChargeParameterDiscoveryRes with “ResponseCode =
OK” and "EVSEProcessing=Finished" is CableCheckReq. Therefore the WHITE Beet is following ISO15118-2 requirement [V2G2-538] here, that is indicating a session error.

I do not see any obvious reason in the logs for this behavior, the session is only allowed to be closed due to some kind of emergency or timeout (timeouts looking good) at this point in the communication, so the PEV is violating the standard here. Do you see any additional information on the PEV side? I assume this is an actual charger where power is also transferred?

Best regards,
lho

kwoyhope, 2024-05-31:
Hello, I'm a SECC SW developer(not SevenStax), and I'd like to share some comments what I know.
There're two weird(seems wrong) points in the 'ChargeParameterDiscoveryRes',
(1) .SAScheduleList.SAScheduleTupleSAScheduleTupleID == 0
-> it should be '1', ('0' is not allowed value)
> in DIN, SAScheduleTupleSAScheduleTupleID == SAIDType(EXI schema "xs:short")
in ISO-2, SAScheduleTupleSAScheduleTupleID  == SAIDType(EXI schema "xs:unsignedByte" min=1, max=255

(2) .DC_EVSEChargeParameter.DC_EVSEStatus.EVSEStatusCode == EVSE_NotReady
-> it's better(strongly recommend) to be 'EVSE_Ready'
> [V2G-DC-347] Table 68 - Semantics and type definition for EVSEStatusCodeType
EVSE_NotReady : Not authorized, StandBy, on maintenance, …
EVSE_Ready : Standard value used during normal operation

lho-stx, 2024-05-31:
Hey @kwoyhope,

thank you for pointing out, I missed this one, please apologize, that was my fault. I will have a look into it and will update you as soon as I have any findings.

Best regards,
lho

nikunjp26, 2024-06-06:
Dear Iho,

any update on this?

best regards,

lho-stx, 2024-06-06:
Hey @nikunjp26,

I found the issue and will create a fix in addition to the one you already got. I will share the file via e-Mail.

Best regards,
lho

vsCodeGB, 2024-07-16:
Hello @lho-stx,  please look at this Wireshark log. Is it the same issue?
I get Failed_SequenceError.
Firmware: Experimental V02_01_00b
HOST: SPI

I have tired with DIN, ISO, DIN+ISO, different EVSE status and voltage/current parameters, each time I get stop session with Failed_SequenceError

https://mega.nz/file/sfEyABbS#q8uIc5TpFvIprWu4Tuo6GcJdFBBWRXkmqEso6Hb06ys

lho-stx, 2024-07-17:
Hey @vsCodeGB,

thank you for the log file. On a first glance it is not the same issue. The SAScheduleID is set correctly, still the PEV fails the charging session after the ChargeParameterResponse. It is also waiting very long to send the SessionStopRequest.

I need to have a closer look into the log file. 

Best regards,
lho

kwoyhope, 2024-07-26:
Hello, I'm a SECC SW developer(not SevenStax), and I'd like to share some comments what I know.
"DC_EVSEStatus.EVSEStatusCode == EVSE_IsolationMonitoringActive"(and IsolationStatus == Invalid, EVSEProcessing == Finished) in your log file(lucid_tries_fail.pcap), but it's (normally) 'EVSE_Ready'(or could be 'EVSE_NotReady') at ChargeParameterDiscovery Stage. ('EVSE_IsolationMonitoringActive' is usually used at CableCheck Stage.)
[image]

nikunjp26, 2024-08-26:
@lho-stx 
Is there any further update on this?

Best regards 
Nikunj Patel