Sevenstax/FreeV2G #322: Spi Data Exchange frame Identifier error after some time

xuanlanghuang, 2024-06-05:
**Before you post**
- [x] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [x] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: PEV
Firmware Version: i.e. V01_00_07
Host Controller Interface: SPI
Host:  Own Implementation

**Describe the bug**
Spi Data Exchange frame Identifier error after some time 

**To Reproduce**
This happen consistently after some time the PEV WB is running, the identifier of data exchange has a 0xD5 at the beginning and the whole message is shifted, this cause our logic to think the WB is out of sync and have an incomplete message
the spi baudrate  is 7,000,000

**Logs**
[image]

[spi-out-of-sync.zip](https://github.com/user-attachments/files/15593044/spi-out-of-sync.zip)
(out of sync at 64.33s for the "wb_outofsync-64-33s" , and 2.188s for "wb_outofsync2-2-188s")

**Additional context**
Add any other context about the problem here.

lho-stx, 2024-06-14:
Hey @xuanlanghuang,

I tried to reproduce your issue, but had no luck. Are you using the WHITE Beet carrier board or do you have a custom Hardware? If you are using WHITE Beet PI carrier board, what is your host device? Do you experience glitches in any other part of the communication? I want to rule out the possibility that it is a problem related to the hardware (e.g. distorted signals). But if this is happening in the setup bytes only, then I need to investigate further. Maybe I'll send you a special debug firmware, because I am not able to reproduce the issue.

Best regards,
lho

lho-stx, 2024-06-14:
Hey @xuanlanghuang,

I checked the logs again and had a look into the timing. Between the CS going low and the first transmission are 10.5μs. In the WHITE Beet manual chapter 9.2.6 the minimum delay time is 20μs.

[image]

Please adapt your host implementation to wait at least 20μs before starting data transmission after asserting the CS line.

Best regards,
lho

xuanlanghuang, 2024-07-17:
Hi, 

I managed to add the delay before the spi send.

But during our testing, we found an edge case where the tx-pending go up during a request size exchange and data exchange, this will cause the WB to not have a zero size exchange.
[image]

What is the correct way to handle this situation?
[txpendingHigh-1ms.zip](https://github.com/user-attachments/files/16274847/txpendingHigh-1ms.zip)

Thanks you.

lho-stx, 2024-07-18:
Hey @xuanlanghuang,

I will check the log files!

Best regards,
lho

xuanlanghuang, 2024-07-29:
Is info@sevenstax.de the right contact email? I don't see a contact form from the website.

Thanks,

lho-stx, 2024-07-29:
Yes!

lho-stx, 2024-08-06:
Closing this issue. We will work on it with the customer directly!

closed 2024-08-06 by lho-stx