Sevenstax/FreeV2G #323: Session was not going into charging state

nikunjp26, 2024-06-06:
Platform: EVSE
Firmware Version: i.e. V02_01_00 with retransition patch
Host Controller Interface: Ethernet
Host: Own Implementation

Dear Team

We are at bus depo to capture the uart logs of why the charging session stop but yesterday night we didn't get that issue.

but we are able  to capture the logs for "Session was not going into charging state" one time. please find the pcap and uart logs file.
 
[teraterm02.zip](https://github.com/user-attachments/files/15605352/teraterm02.zip)

[Uploading umc01.zip…]()

**Refer from line number - 2329 from "teraterm02" file
Refer around 417247 packet number from "umc01.pcapng" file**

FYI :  we are also here for today also if you need anything test so you can guide us.

Best Regards,
Nikunj Patel

nikunjp26, 2024-06-06:
Dear Team,

[Uploading umc01.zip…]()

please find again

nikunjp26, 2024-06-06:
Dear Iho,
Please find the attached file

[umc01.zip](https://github.com/user-attachments/files/15667316/umc01.zip)

nikunjp26, 2024-06-12:
Dear Iho,

We capture the same scenario again so please find the uart and wireshark logs.

[image]
[error before charging sesion.zip](https://github.com/user-attachments/files/15805322/error.before.charging.sesion.zip)

Please check the logs as per the attached snapshot and support to solve the issue.

Best Regards,

lho-stx, 2024-06-13:
Hey,

multiple charging sessions. After a quick glance there are the same symptoms as in #326 and #327.

Best regards,
lho

closed 2024-07-05 by github-actions[bot]