Sevenstax/FreeV2G #330: SPI_TX_PENDING is always high 3,3V

AdrianKn, 2024-07-03:
**Before you post**
- [ ] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [ ] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: EVSE
Firmware Version: dont know
Host Controller Interface: SPI
Host: Own Implementation

**Describe the bug**
I am using a White Beet Carrier Board EI 1.1 and would like to control it via SPI using a SAMV71Q21B microcontroller.
I have the following problem the pin SPI_TX_PENDING is permanently high (3.3V) when I try to read out the White Beet, it responds to me in the Size Exchange Frame with aa aa 00 00 which means that it cannot send me any messages (Figure 5). What could be the reason that the SPI_TX_PENDING is not set to low? 

As you can see in the next 4 oscilloscope images (Figure 1-4), the communication with the configuration commands and the corresponding responses works. 

**To Reproduce**
Connect the white beet to the power supply and the pin is directly high, when I read the size exchange frame I get a size of zero back and the pin is still high.
The pin is also high if I don't have it connected to my host controller and my host controller pin is set as input and has 0V, do I need to use a pull down resistor?

**Expected behavior**
I would expect that when I send a size exchange frame, it would send me back aa aa and a size and not a size of zero when there is a status message to send. I think when I read the message the SPI_TX_PENDING pin should go low, but it doesn't. I was expecting the pin to be low when the device starts up if there is no message or if there is one after i read it.

**Logs**
I can not create a Log because i send nothing.

**Additional context**
Add any other context about the problem here.

lho-stx, 2024-07-03:
Hey @AdrianKn,

thanks for creating the new issue. After a reset the WHITE Beet sends a notification to the host, so the TX pending line is expected to go high, but with a short delay after the reset **and** you should see a payload size in the size exchange frame. If the pin is high immediately you might have a faulty board. You do not need a pull down resistor.

Unfortunately I can not help you further, if you can not create log files of a supported file format (please have another look at the [guide how to create log files](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files#spi)). 

There is an issue which discusses how to setup the SPI communication between custom host controller and WHITE Beet. [This comment](https://github.com/Sevenstax/FreeV2G/issues/150#issuecomment-1437081690) has more information and also a log file with example SPI traffic between host and WHITE BEet. Maybe you can get some insights from the log file.

Does the board work with the [recommended Raspberry Pi 4 setup](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#use-case-spi-raspberrypi-example)? If not, then you most likely have have faulty board, please contact your Codico sales representative. If yes, then we need to investigate your setup further.

Best regards,
lho

AdrianKn, 2024-07-03:
Thank you very much for your prompt and professional assistance. 
I have already come to the same conclusion that the board is faulty. Due to the reset the SPI_TX_PENDING pin is still high. We will contact Codico and get a new White Beet Carrier Board as soon as possible and try again with the new one. Thanks for your help : )

AdrianKn, 2024-07-03:
One last question, if I do a factory reset on the carrier board, will it be in the same condition as when I received it? Or do I need to make any special adjustments? So far I have only tried the normal reset command and the reset button.

lho-stx, 2024-07-04:
Hey @AdrianKn,

Yes, this should not happen at all.
If you did not upload any certificates for using the Plug and Charge feature yet, resetting the board is enough.

Great to see you found the issue quickly. I'll close the issue.

Best regards,
lho

closed 2024-07-04 by lho-stx