Sevenstax/FreeV2G #332: Session still stopped after ServiceDetailReq

Kristanpivk123, 2024-07-14:
**Before you post**
- [x] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [x] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: PEV evaluation board
Firmware Version: V01_00_07
Host Controller Interface: SPI
Host: Own Implementation

**Describe the bug**
On two of four EVSE that I tried to comumunicate with, the session was stopped after EVSE didn't accept ServiceDetailReq message from WHITE Beet PEV.
This but was already reported multiple times (issues #76  #92  #142), to all of which you answered that it will be fixed in the upcoming FW update. I am currently using the latest FW but this is apparently still not fixed.

**To Reproduce**
Use an EVSE from either Alpitronic or Delta and try to establish DC charging with it

**Expected behavior**
According to Google research and what was mentioned on the previous issues, this is not mandatory for ISO 15118 so WHITE Beet should be able to skip this or at least there should be an option in V2G configuration to skip this step.

**Logs**
I am attaching UART logs of both EVSE where this happened (files differ a bit from the original logs due to my own implenetation in capturing them):
[alpitronic_test   1 dbg.txt](https://github.com/user-attachments/files/16227884/alpitronic_test.1.dbg.txt)
[delta_test   1 dbg.txt](https://github.com/user-attachments/files/16227885/delta_test.1.dbg.txt)

**Additional context**
This week I am going to try to get around this issue by using DIN 70121 which should probably work (according to other issues) but it would be great if this was supported. So far i tested 4 EVSE with my SPI implementation. 2 were using DIN 70121 and two were using ISO 15118. Both that used ISO were terminated by the above mentioned problem. 

Tommorow I plan to test another EVSE which uses Vector SECC. According to other issues I also expect this same problem there. If that is true, that would mean that pratically most public DC EVSE (in Slovenia) don't support ServiceDetailReq which is quite a big problem, especially if they also don't support DIN 70121 (no way to connect to them in that case).

I also must mention another test of DC EVSE with ISO 15118 that worked. I executed that test with FreeV2G (at that time I still didn't have my own SPI implementation finished). It was on an EVSE that used EcoG SECC. At the time I was still using V01_00_06 from the summer 2023 (version that was bricked for updates so I had to send it to you to programm ti with V01_00_07). If needed I can provide Wireshark log of this communication with mirroring. I added this to give you additional informations regarding this bug fix (in case that it was already fixed for V10_00_06 but was overwritten for V01_00_07).

lho-stx, 2024-07-15:
Hey @Kristanpivk123,

we are working on the next WHITE Beet PI release at the moment. I can not give you an official release yet, but it should be available soon. Nevertheless, sharing a log would be great.

Best regards,
lho

Kristanpivk123, 2024-11-08:
Hey Iho,

hope that the update for WHITE Beet PI is going well. Would you be able to estimate when this new release will be available?
It's important for me to know so I can estimate our project's timeline as this is something that absolutelly needs to be fixed if we wish to complete the project.

Best regards,
Kristan

lho-stx, 2024-11-12:
Hey @Kristanpivk123,

the update is already available at CODICOs download area. Please have a look there or ask your CODICO represantive.

Best regards,
lho

closed 2024-11-12 by lho-stx