Sevenstax/FreeV2G #333: Getting Unexpected Sub ID while testing EVSE and EV simulator with V2.1.0 firmware

jptalledo10, 2024-07-16:
**Before you post**
- [ X] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [ X] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: EVSE
Firmware Version: V02_01_0
Host Controller Interface: Ethernet 
Host: FreeV2G (EVSE_v2.0.1_4)

**Describe the bug**
When using two white beet boards for EVSE <-> EV, I am getting an error indicating:
```
  File "/home/evbuddy/github/FreeV2G/Whitebeet.py", line 175, in _sendReceive
    raise Warning("Response from mod ID {:02X} with unexpected sub ID {:02X} received".format(response.mod_id, response.sub_id))
Warning: Response from mod ID 29 with unexpected sub ID 81 received
```

**To Reproduce**
Running the following command:
sudo python3 Application.py eth -m c4:93:00:34:ab:03 -r EVSE -i enp0s31f6 -c ev.json

**Expected behavior**
As Firmware 1.7.0 showed in the past, a full charging sequence should be expected

**Logs**

**Additional context**

lho-stx, 2024-07-17:
Hey @jptalledo10,

you used the EV configuration file but addressed a WHITE Beet EI. Please try:

```bash
sudo python3 Application.py eth -m c4:93:00:34:ab:03 -r EVSE -i enp0s31f6
```

Best regards,
lho

lho-stx, 2024-07-23:
Hey,

often a reason for failing sessions are non matching configurations of simulator and WHITE Beet. Please check that your configs match (e.g. choose ISO and DC_extended only).

If this is not resolving your issue, please create the wireshark log files as asked in the guide and post them here, they are mandatory for resolving issues.

Best regards,
lho

jptalledo10, 2024-07-24:
@lho-stx  

Can you advise where to check for ISO/DIN configurations. On the EVSE I have the following:

`{
    "mac": "00:01:01:63:77:31",
    "ev": {
        "evid": "00:01:01:63:77:31",
        "protocol_count": 1,
        "protocols": [
            0
        ],
        "payment_method_count": 1,
        "payment_method": [
            0
        ],
        "energy_transfer_mode_count": 1,
        "energy_transfer_mode": [
            1
        ]
    },
    "battery": {
        "timestep": 10000,
        "capacity": 50000,
        "level": 49000,
        "full_soc": 100,
        "bulk_soc": 80,
        "max_current": 100,
        "max_power": 12000,
        "max_voltage": 300,
        "target_current": 40,
        "target_voltage": 200,
        "target_voltage_delta": 10,
        "max_current_AC": 0,
        "max_voltage_AC": 0,
        "min_current_AC": 0
    }
}`

Attaching the Wireshark trace captured on on EVSE<-> EV Simulator session
[evse.zip](https://github.com/user-attachments/files/16356903/evse.zip)

lho-stx, 2024-07-25:
Hey @jptalledo10,

the values in the json file are the values found in the WHITE Beet manual. For supporting DIN and ISO you need to set 

```json
"protocol_count": 1, "protocols": [ 0, 1 ]
```

In the EV simulator you need to set the protocols in the V2G module and the energy transfer mode in the battery module.

Will check the log files!

BR,
lho

lho-stx, 2024-07-25:
Hey @jptalledo10,

unfortunately there are no V2G messages seen in the log file. Most likely the port mirror is not enabled. Can you please enable the port mirror (taken from the how to [create log files guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files)):

> If you are using FreeV2G the port mirror can be enabled by starting Application.py with the -p flag.

and recreate the log file?

Best regards,
lho

jptalledo10, 2024-07-25:
@lho-stx 

Using the EVSE_v2.0.1_4 branch. the -p argument is not accepted by the Application.py. Can you advise another method to enable the port mirror feature

[image]

lho-stx, 2024-07-26:
Hey,

I missed that you are still on the old firmware, my fault, I am sorry!

You need to insert 

```
evse.whitebeet.networkConfigSetPortMirrorState(1)
```

after [line 102 in Application.py](https://github.com/Sevenstax/FreeV2G/blob/d59d38949463c5ba1b28fe51cf2eb39330e643d1/Application.py#L102) in order to enable the port mirror.

Please also consider updating the firmware, so you can use the latest commit and have all bug fixes in the firmware.

Best regards,
lho

jptalledo10, 2024-07-27:
Thanks for the clarification. Here is the Wireshark log and the snapshot of the EV Simulator

[image]

Here is the log running v2.0.1_4
[evse1.zip](https://github.com/user-attachments/files/16398384/evse1.zip)

Changing to the EVSE_v2.1.0_1 worked and able to run the full loop including authentication/confirmation

lho-stx, 2024-07-29:
Hey @jptalledo10,

great to see your setup is working now! I will close this ticket.

Best regards,
lho

closed 2024-07-29 by lho-stx