Sevenstax/FreeV2G #334: EV - FW V01_00_07 - Port mirroring - CM_ATTEN_CHAR.IND not forwarded

giguerator, 2024-07-23:
**Before you post**
- [ ] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [ ] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: PEV
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet / SPI
Host: FreeV2G (EV_v1.0.7_0) - Slightly modified to add J1772 basic communication (disable slac) and add additional timeouts

**Describe the bug**

**To Reproduce**
Launch an EV session with FreeV2G with port mirroring activated
-> Application.py eth -i Ethernet -m C4:93:00:34:D0:77 -r EV -t 60 -p

**Expected behavior**
When capturing the Ethernet packets between the whitebeet and the host computer on Wireshark, I want to see the CM_ATTEN_CHAR.IND from the EVSE but I only see messages from the whitebeet to the EVSE

**Logs**
See wireshark capture below (remove .txt extension).
[ccs_120724.pcapng.txt](https://github.com/user-attachments/files/16349049/ccs_120724.pcapng.txt)

I don't currently have access to the whitebeet usb port without modifying the equipment in which the whitebeet is installed.

**Additional context**
Add any other context about the problem here.

lho-stx, 2024-07-25:
Hi @giguerator,

I checked your log and I can reproduce your issue in my setup. When enabling the port mirror I see all the expected messages. 

Is the charging session working? Because I also see no V2G traffic in your logs. I see the framing message that is enabling port mirror, but no traffic I would expect when the port mirror is enabled. I see that SLAC is successful and also SECCDP requests. This behavior might be seen when a network switch is placed in between your host and the WHITE Beet. The network switch might filter out non-broadcast messages, even with port mirror enabled. This would explain such behavior.

Do you have network gear placed between WHITE Beet and host?

Best regards.
lho

giguerator, 2024-07-25:
Hi @lho-stx ,

If that's the issue you are making me feel a bit dumb for not thinking of it...!

I have an unmanaged switch between the whitebeet and the computer. The switch has broadcast storm protection capability, but it's disable. It's totally possible that there are other filtering that is not explained in the user manual.

As the other messages are also broadcast, I never suspected the switch.

I will test connecting the whitebeet directly to the host on next monday and reply here with the results.

Benjamin

lho-stx, 2024-07-26:
Hey @giguerator,

no problem! 

You can get a switch working by using VLANs and virtual network interfaces on your host. This is not officially supported (yet), but I had success with this setup. You basically need to create a tagged VLAN for the WHITE Beet and a corresponding interface on your host. If you have multiple WHITE Beets connected this way, you need to make sure that each virtual interface has its own MAC address.

Please let me know if removing the switch is resolving your issue.

Best regards,
lho

closed 2024-08-06 by lho-stx