Sevenstax/FreeV2G #335: Question: Firmware Update Problems

DeividasPalu, 2024-07-31:
Hello,

We are currently working with the Whitebeet and its integration into our system. We are using WHITE-beet-EI 1.1, and our goal is to implement the EVSE side.

Initially, the firmware version of our development Whitebeet devices was V01_01_07 (as reported by the Get Firmware Version command via SPI). We successfully updated one of our Whitebeet units to version V02_00_01 and another unit to version V02_01_00 using the firmware update module. 
After the update was completed, both of the Whitebeet devices report the new firmware version, are responsive, acknowledging commands, generating CP signals, and performing similar tasks.

However, we have encountered issues with further firmware updates. We want to ensure that we can update the Whitebeet devices at any time, but currently, after the initial update, the devices refuse further updates. They accept packets up to approximately 1.3 kB, after which the Firmware Update Module returns a status message indicating "Error was detected" (with module ID 0x11 and sub ID 0x82). Even the one with older version (V02_00_01) do not let us update firmware again to the newest version (V02_01_00).

Can the firmware be upgraded more than once? Can it be downgraded? What could be the issue if the Whitebeet Firmware Update Module is not working properly?

To help clarify the situation, I have added some screenshots:

We are sending packets to the Whitebeet one by one, and each packet is acknowledged. However, after 1280 bytes, we receive a status frame with an error message.
[image]

Debug information from whitebeet via UART :
[image]
[image]

lho-stx, 2024-07-31:
Hey @DeividasPalu,

I am sorry that you are experiencing such issues. Of cause the firmware can be upgraded more then once. Unfortunately you used the wrong issue template, so you did not see that we need to have log files of your issues. Can you please follow [this guide for creating log files](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files), so we can inspect and try to reproduce your issue? We need SPI and UART logs in your case.

I kindly ask you to fill in this table to give an overview of the scenarios you tested:

|   from \ to    | 1.1.7 | 2.0.1 | 2.1.0 |
| ----- | ----- | ----- | ----- |
| 1.1.7 |       |    working   |  working     |
| 2.0.1 |       |       |   not working    |
| 2.1.0 |       |       |       |

Best regards,
lho

DeividasPalu, 2024-07-31:
> Hey @DeividasPalu,
>
> I am sorry that you are experiencing such issues. Of cause the firmware can be upgraded more then once. Unfortunately you used the wrong issue template, so you did not see that [...]

What has happened?

Whitebeet device #1: 
      1. Update 1.1.7 -> 2.1.0 (success)
      2. Try Update 2.1.0 -> 2.1.0 (failure)
      3. Try Downgrade 2.1.0 -> 2.0.1 (failure)
      4. Try Downgrade 2.1.0 -> 1.1.7 (failure)

Whitebeet device #2: 
      1. Update 1.1.7 -> 2.0.1 (success)
      2. Try Update 2.0.1 -> 2.1.0 (failure)
      3. Try Update 2.0.1 -> 2.0.1 (failure)
      4. Try Downgrade 2.0.1 -> 1.1.7 (failure)

Here is the raw SPI communication dump, to follow up with the uart log i've included previously:

`SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 09
SPI_DMA -> send (0013 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0013 bytes) = 55 55 00 00 C0 05 81 FF 00 01 01 F4 C1
SPI_DMA -> send (0004 bytes) = AA AA 00 0C
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0016 bytes) = 55 55 00 00 C0 0E 00 EE 00 04 BE BA FE CA 39 C1
SPI_DMA -> recv (0016 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 0C
SPI_DMA -> send (0016 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0016 bytes) = 55 55 00 00 C0 0E 01 EE 00 04 BE BA FE CA 38 C1
SPI_DMA -> send (0004 bytes) = AA AA 00 08
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0012 bytes) = 55 55 00 00 C0 10 40 01 00 00 2C C1
SPI_DMA -> recv (0012 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 11
SPI_DMA -> send (0021 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0021 bytes) = 55 55 00 00 C0 10 40 01 00 09 00 07 30 2E 30 2E 30 2E 30 D0 C1
SPI_DMA -> send (0004 bytes) = AA AA 00 08
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0012 bytes) = 55 55 00 00 C0 10 41 02 00 00 2A C1
SPI_DMA -> recv (0012 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 13
SPI_DMA -> send (0023 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0023 bytes) = 55 55 00 00 C0 10 41 02 00 0B 00 09 56 30 32 5F 30 31 5F 30 30 DC C1
SPI_DMA -> send (0004 bytes) = AA AA 00 09
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0013 bytes) = 55 55 00 00 C0 11 41 02 00 01 02 26 C1
SPI_DMA -> recv (0013 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 0B
SPI_DMA -> send (0015 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0015 bytes) = 55 55 00 00 C0 11 41 02 00 03 00 10 48 CD C1
SPI_DMA -> send (0004 bytes) = AA AA 00 8C
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0144 bytes) = 55 55 00 00 C0 11 04 00 00 84 00 00 00 00 53 54 58 46 01 01 00 32 00 07 EB E3 07 D0 00 02 02 01 42 71 F4 B5 EF 39 F6 BD 3E CB A9 0D 03 CE 0C 83 F9 B3 E5 19 F5 66 1C 20 28 02 39 47 6A 96 A5 24 AA 55 00 2F 77 33 01 00 00 00 00 25 00 00 10 00 0E 43 53 45 56 45 4E 53 54 41 58 20 47 6D 62 48 11 00 0F 43 49 53 4F 31 35 31 31 38 20 28 45 56 53 45 29 B6 E1 4D F2 5D 32 C1 91 5B 73 62 7E 66 6B 76 A3 2C E5 3C B6 B6 6B E7 72 3E BA D5 5A C1
SPI_DMA -> recv (0144 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 09
SPI_DMA -> send (0013 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0013 bytes) = 55 55 00 00 C0 11 04 00 00 01 00 67 C1
SPI_DMA -> send (0004 bytes) = AA AA 00 8C
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
SPI_DMA -> send (0144 bytes) = 55 55 00 00 C0 11 04 00 00 84 00 00 00 01 BE 40 C6 44 C2 AA 55 04 53 77 33 05 03 00 00 00 02 00 00 02 00 77 33 05 01 00 00 04 3D 00 05 00 00 02 01 00 00 01 00 39 BB 09 7B E9 08 9F DB 22 E2 74 D5 34 59 13 A5 8B BA E8 4B E6 D8 E8 E5 FD D6 0E 5C EA 50 99 A3 97 32 21 27 CC B5 25 8A C1 B8 B5 B4 91 28 B2 A0 6F 56 AF 15 4E 4C 0D CA AE BA CD 83 AE 72 0B 66 23 67 A7 55 6E 2F 56 1A 56 41 A8 8C CC B1 E8 F2 4A 82 E9 FE 47 0D 47 8E A9 09 C1
SPI_DMA -> recv (0144 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 09
[... 65 lines cut ...]
SPI_DMA -> recv (0013 bytes) = 55 55 00 00 C0 11 04 00 00 01 00 67 C1
SPI_DMA -> send (0004 bytes) = AA AA 00 8C
SPI_DMA -> recv (0004 bytes) = AA AA 00 00
[... cut ...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> send (0004 bytes) = AA AA 00 00
SPI_DMA -> recv (0004 bytes) = AA AA 00 09
SPI_DMA -> send (0013 bytes) = 55 55 00 00 00 00 00 00 00 00 00 00 00
SPI_DMA -> recv (0013 bytes) = 55 55 00 00 C0 11 04 00 00 01 06 61 C1
`

Please note that we have reduced the packet size to 128 bytes in order to try and see if the device takes any data at all. 
(Which it does)

lho-stx, 2024-08-13:
Hey @DeividasPalu,

we tried your use cases, but did not see this issue. Did you set the time before running the update?

We need some more time for the investigation of the log file.

Best regards,
lho

DeividasPalu, 2024-08-19:
To clarify, are you referring to setting the time through the Time Module? I did not perform the "Set UTC Time" operation.
[image]

lho-stx, 2024-08-19:
Hey,

yes! Please try setting the UTC time before running the update script.

Best regards,
lho

DeividasPalu, 2024-08-22:
hello,

I've been working on updating the modules and tried setting the time before the update. Version 1.1.7 has no integrated time module. Whenever I attempt to set the time, I receive the same message:
55 55 00 00 C0 FF F4 01 00 00 87 C1 00 00 00 00  (I assume it means "modules does not exist" )
[image]

despite that, I can always successfully update from version 1.1.7:
1. Update 1.1.7 -> 2.0.1 (success)
2. Update 1.1.7 -> 2.1.0 (success)

Time module helped with the newest version - 2.1.0, After setting up the time, I'm able to downgrade from version 2.1.0 successfully:
3. Downgrade 2.1.0 -> 2.0.1 (success)
4. Downgrade 2.1.0 -> 1.1.7 (success)

However, I'm still encountering the same issue where the update gets stuck around 1240 bytes when trying to update or downgrade from version 2.0.1:
5.  Downgrade 2.0.1 -> 1.1.7 (failed)
6.  Update  2.0.1 -> 2.1.0 (failed)

Is there any other modules that need to be activated, or something similar, for this version ?

lho-stx, 2024-08-26:
Hey,

the downgrade from 2.0.1 to 1.1.7 may fail, because there was a change in the filesystem of the WHITE Beet. The 2.0.1 and 1.1.7 filesystems are not compatible. You may try to downgrade the filesystem first and then the firmware. But this is not officially supported, we only support the latest firmware version.

What I don't understand is the failing update from 2.0.1 to 2.1.0. Did you try to update the firmware only or the firmware AND the filesystem? You just need to update the firmware. The filesystem update is only needed when updating from version 1.1.7.

Please try the update 2.0.1 to 2.1.0 with the firmware only and tell me if this works. Please also make sure to use the correct tool, the 2.X.X firmware updates need the new version of the firmware update tool, that is setting the time.

Best regards,
lho

Best regards,
lho