Sevenstax/FreeV2G #337: Session hanging after SLAC

giguerator, 2024-08-08:
**Before you post**
- [x] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [ x] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Setup**
Platform: PEV 
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.7_0) / Modified

**Describe the bug**
It may not be a bug, but I really don't understand what is happening. I mounted a Trialog CCS sniffer (https://www.trialog.com/wp-content/uploads/2020/10/EV-ComboCS_CCS-Tester_Datasheet_v5.pdf) between the Whitebeet and the EVSE and since then, SLAC is taking very long and something wrong is happening in the network establishment after that.

I have swapped all the wiring between the Whitebeet and the EVSE with no change in the behavior. I tried it with the FreeV2G development branch without my modifications. I swapped the Whitebeet with another one (that was also used with the trialog though...).

I need help to figure out if the Whitebeet is damage and/or what are the next troubleshooting steps. I joined both the uart logs and the wireshark capture from port forwarding to the issue.

**To Reproduce**
Start a charging session...

**Expected behavior**
After SLAC, network establishment occurs and the session starts.
[logs.txt](https://github.com/user-attachments/files/16548165/logs.txt)
[SLACCapture.pcapng.txt](https://github.com/user-attachments/files/16548170/SLACCapture.pcapng.txt)

**Logs**
APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_07
  Target:    WHITE-BEET
  CPU:       STM32F745

PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_03_00
 Hex:    0x000a0300
 Dec:    100300

 Hostcontroller Interface: ETHERNET!

APPLICATION RUNNING ...
[    0.009] [NOTICE] SLAC: stxSLAC_Init()
[    0.010] [NOTICE] APL-ETH:   MAC of NIC[0]:  c4:93:00:34:cf:c4
[    2.303] [NOTICE] APL-ETH:   MAC of NIC[1]:  c4:93:00:34:cf:c3
[    2.303] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    2.304] [NOTICE] APL-MAIN: BOOT-CFG: FW1
[    2.305] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EV
[    2.306] [NOTICE] PLCV-HL:  Boot from host selected
[    2.306] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    2.316] [NOTICE] APL-ETH: Got link:
[    2.316] [NOTICE] APL-ETH:     Interface: eth1
[    2.317] [NOTICE] APL-ETH:     MAC:       C4:93:00:34:CF:C3
[    2.522] [NOTICE] PLCV-HL: Reset QCA700x
[    3.387] [NOTICE] PLCV-HL:  =======================================
[    3.388] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[    3.903] [NOTICE] PLCV-HL:  =======================================
[    3.903] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[    3.911] [NOTICE] APL-MAIN:
[    3.911] [NOTICE] APL-MAIN: =============================
[    3.912] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6):
[    3.912] [NOTICE] APL-MAIN: for IP Config handle [fe01]
[    3.913] [NOTICE] APL-MAIN: IPv6 Addresses:
[    3.914] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe34:cfc3
[    3.914] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[    3.915] [NOTICE] APL-MAIN:   DNS1:  2001:4860:4860:0 : 0:0:0:8888
[    3.916] [NOTICE] APL-MAIN:   DNS2:  fd00:0:0:0 : be05:43ff:fe44:5280
[    3.917] [NOTICE] APL-MAIN: =============================
[    3.918] [NOTICE] APL-MAIN:
[    4.045] [NOTICE] PLC-HL:  Own PLC MAC = C4:93:00:34:CF:C5
[   11.919] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365
[   94.027] [NOTICE] SLAC: stxSLAC_Stop()
[... 808 lines cut ...]
[  942.230] [NOTICE] APL-MAIN: =============================
[  942.230] [NOTICE] APL-MAIN:
[  944.107] [NOTICE] PLCV-HL:  =======================================
[  944.108] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  944.625] [NOTICE] PLCV-HL:  =======================================
[  944.626] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[  948.238] [NOTICE] PLCV-HL:  =======================================
[  948.239] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  948.756] [NOTICE] PLCV-HL:  =======================================
[  948.757] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[  952.371] [NOTICE] PLCV-HL:  =======================================
[  952.372] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  952.889] [NOTICE] PLCV-HL:  =======================================
[  952.890] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[  956.505] [NOTICE] PLCV-HL:  =======================================
[  956.506] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  957.023] [NOTICE] PLCV-HL:  =======================================
[  957.024] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[  960.639] [NOTICE] PLCV-HL:  =======================================
[  960.640] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  961.157] [NOTICE] PLCV-HL:  =======================================
[  961.158] [NOTICE] PLCV-HL:    BOOT FROM HOST READY

**Additional context**
Add any other context about the problem here.

giguerator, 2024-08-13:
Is it normal to have the PLV-HL repeated messages?

lho-stx, 2024-08-14:
Hey @giguerator,

I am also a bit confused. The PLCV-HL messages show, that the QCA is somehow reset and the firmware is uploaded again. I also see the message  (this matches also the pcapg file):

```
V2GCPS: Duty cycle is 0 permill
```

After this the data link on the plc interface is dropped and the PLCV-HL messages are spamming the log file. Duty cycle 0 means short between CP and PE (basically an error case). I think either your communication peer is not acting accordingly to ISO/IEC or your board is damaged. I don't think your board is damaged, because I also see

```
V2GCPS: Duty cycle is 55 permill
```

so it is reading the duty cycle correctly at some point.

I would need to test this scenario to check if I get similar log files, when I set the CP state to state E after SLAC matching.

Does this issue also appears when using another peer then the CCS sniffer?

Best regards,
lho

giguerator, 2024-08-14:
Hi @lho-stx , a quick clarification. This issue started happening after I installed the Trialog sniffer for a day between the whitebeet and my EVSE. I removed the sniffer and the situation remains. 

Regarding the "V2GCPS: Duty cycle is 0 permill". You can see that it happens about 40s after the 55 permil (I just noticed that you have 2 sessions in the UART logs, but only one in the PCAP...). The 0 permil happens after the EVSE gives the error "Too many errors occurred, stopping communication". The EVSE is stopping its PWM.

lho-stx, 2024-08-16:
Hi,

I will recheck the log files, it did not see any problems yet. This might hint that there is indeed a HW problem. Can you tell what error the EVSE sees specifically?

Best regards,
lho

giguerator, 2024-08-16:
Hi @lho-stx, I found the issue yesterday. I don't know the root cause but it was linked to the PCI express NIC the Whitebeet was connected to. I swapped the Whitebeet connection with the facility LAN on the computer and the issue disappeared.

I noticed that I could see some messages, such as the SLAC acknowledge message, from the Whitebeet to the host computer on Wireshark. However, the messages were not see by FreeV2G so the code was getting stuck.

Fyi I was using a Realted RTL8168 NIC.

lho-stx, 2024-08-16:
Hey @giguerator,

maybe the PCI express NIC is acting as a sort of network switch/router? We know using WHITE Beet with routers and switches need some setup in order to work correctly (basically using VLANs in order to be sure the broadcast messages are send out correctly).

Thank you for sharing the information on the NIC used, highly appreciated.

If you issues are resolved completly, please feel free to close this issue ticket.

Best regards,
lho

giguerator, 2024-08-16:
@lho-stx I am closing the ticket.

The messages that I saw on wireshark and that did not make their way to FreeV2G were not broadcast messages though. They were the non-standard messages from your host-whitebeet API. The issue was mainly around the SLAC ack, but sometimes during the network establishment. The other weird thing is that if I started the FreeV2G session, stopped it while it was waiting for an EVSE to connect with CTRL+C and started it again without removing power to the whitebeet, it would work.

giguerator, 2025-04-28:
@lho-stx we found the issue and browsing the issues list it seems quite a few people are having the same issue. The issue is that for some Network Interface Cards, the Nagle's algorithm is delaying the application notification for the SLAC matched packet to bundle it with the next packet. In our case, we could see the SLAC matched message clearly in at the expected time in Wireshark but then, we could also clearly see that the application did not process the SLAC matched until it got the next message which was a CP change message generated by the EVSE timing out. We disabled Nagle's algorithm for that interface (in Windows Registry) and now everything's running perfect. I suggest to deactivate Nagle's as part of the Python code.