Sevenstax/FreeV2G #338: Question: Codico Whitebeet 1.1 Ethernet - How to Permanently Disable Interface?

brunesi, 2024-08-09:
Hello,

We are currently integrating the Codico Whitebeet 1.1 with OTA firmware updates into our system. Our setup involves using the board in SPI mode for regular operations and the Ethernet port exclusively for firmware updates. The Ethernet cable remains permanently connected to the board.

In our application, we require a method to deactivate and maintain the Ethernet interface in a down state, as the port should not be active during normal operations. However, despite taking the necessary OS-level steps on our Ubuntu 16.04 system (running on an IPC board), the Ethernet interface keeps reactivating.

We are aware that the github post documentation suggests using Debian 12 for reproducing Ethernet-related scenarios. However, since our application runs on Ubuntu 16.04, we are unable to follow this recommendation. We suspect the issue might be related to the Whitebeet’s internal operations or a possible configuration/hardware setting.

Could you assist us in identifying the necessary steps or configurations to permanently disable the Ethernet interface when not in use?

Thank you for your support.

Marchesi

lho-stx, 2024-08-14:
Hey @brunesi,

you can try disabling the interface by getting the interface information (WHITE Beet manual chapter 13.3.7 Get Interface Info), select the correct interface index (it should be named eth0) and then setting its state to off (WHITE Beet manual chapter 13.3.8 Set Interface State).

Best regards,
lho

brunesi, 2024-08-14:
We tried that today, thank you.
The interface (eth0 on index 1) indeed goes down, however for only about two seconds, then goes up again.

Additionally, we noted that whenever the eth cable is pulled out and pushed back in, the fw messages those events out. So, it appears that there's an instance on the current Sevenstax fw that stays aware of the Ethernet port status. This shouldn't happen, we wonder, if the eth was indeed 'off'.

We are reporting this back as a comment, however we could provide this info with detailed logs and messages, but we hope this might suffice in order to advance this difficulty we're facing right now.

Regards.

lho-stx, 2024-08-16:
Hey,

I was not able reproduce the first issue. Even after setting my interface down and up again I was not able to communicate with the WHITE Beet after setting the interface down.

I need to test replugging the cabee though.

Can you tell me which bytes you send out to disable the interface?

Best regards,
lho

brunesi, 2024-08-16:
Sure, this is a more thorough log:

with eth cable connected
AA AA 00 09 55 55 00 00 C0 05 52 00 00 01 01 24 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 52 00 00 02 00 00 24 C1 <LF>

Getinterface info
AA AA 00 08 55 55 00 00 C0 05 50 00 00 00 28 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 50 00 00 11 00 00 01 04 70 6C 63 30 00 01 01 04 65 74 68 31 00 28 C1 <LF>

eth port leds are ON, with cable connected

SET INTERFACE STATE ON
AA AA 00 0A 55 55 00 00 C0 05 51 00 00 02 01 01 23 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 51 00 00 01 00 26 C1 <LF>

GET INTERFACE STATE
AA AA 00 09 55 55 00 00 C0 05 52 00 00 01 01 24 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 52 00 00 02 00 00 24 C1 <LF>

status is off, although eth port leds are on

SET INTERFACE STATE OFF
leds turn OFF for ~two seconds, then they come up again
AA AA 00 0A 55 55 00 00 C0 05 51 00 00 02 01 00 24 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 51 00 00 01 00 26 C1 <LF>

with leds on, GET STATE command
AA AA 00 09 55 55 00 00 C0 05 52 00 00 01 01 24 C1 <LF>
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 52 00 00 02 00 00 24 C1 <LF>

eth cable being pulled out:
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 80 FF 00 01 01 F5 C1 <LF>
Station went down<LF>

eth cable being pushed in again:
AA AA 00 00 55 55 00 00 00 00 00 00 00 00 00 00 00 55 55 00 00 C0 05 81 FF 00 01 01 F4 C1 <LF>
Station went up<LF>

As you can see, the status reported is not what the hardware is in fact.
Additionaly, some section of the fw is still aware of the cable presence, if the interface were trully off,
we imagine that shouldn't be the case.

Regards.

lho-stx, 2024-08-26:
Hey,

we where not able to reproduce the plug-out/plug-in behavior. In our case the interface kept down. We are still looking into this.

Best regards,
lho

brunesi, 2024-10-15:
Hello,

Any updates on this issue, please?