Sevenstax/FreeV2G #342: Question: what is the proper way to end a charge session?

pxidr, 2024-09-06:
Hello,

I would like to know how to end properly a charge session (after the charge loop), this is unclear on the documentation.

On some cars (like Tesla Model 3/Y) my current solution works fine, but on others (like the Renault Zoe), the CCS connector remains locked on the car, I've to wait almost a minute before the car unlocks it.

There is my current steps (after the charge loop) :
1) Send notification to stop charging (ModId 0x27 SubId 0x75)
2) Send stop charge (ModId 0x27 SubId 0x6F)
3) Open all DC contactors on the charger side
4) Set PWM duty cycle to 100% (ModId 0x29 SubId 0x44)
5) Stop update parameters loop (ModId 0x27 SubId 0x63)
6) Stop listen to HPGP (ModId 0x27 SubId 0x70)
7) Stop SLAC (ModId 0x28 0x43)

Are these steps right?

Thanks, Damien

kwoyhope, 2024-09-13:
Hello, I'm a SECC SW developer(not SevenStax), and I'd like to share some comments what I know.

(1) EVSE's HV DC Contactor should be opened carefully after checking some conditions.
ex) EV agreed to stop charging? (PowerDeliveryReq(stop) received?), HV DC current is enough low(depend on normal/fast/emergency) for safety?, ...

(2) PWM duty should be retained for V2G communication(unless socket closed by peer(EVCC))
> [V2G-DC-672] After the SECC has sent a SessionStopRes message with ResponseCode equal to “OK”, it shall keep the CP oscillator switched on for V2G_SECC_CPOscillator_Retain_Time according to Table 77. The SECC shall then switch the CP oscillator off.

It might be more helpful if attach the log(tcpdump,..) files of test cases(ex, good(?) case of 'Tesla Model 3/Y', bad case of 'Renault Zoe', .. not only for me, but also for others(@lho-stx )).

pxidr, 2024-09-14:
> Hello, I'm a SECC SW developer(not SevenStax), and I'd like to share some comments what I know.
> 
> (1) As you know, EVSE's HV DC Contactor should be opened carefully after checking some conditions. ex) EV agreed to stop charging? (PowerDeliveryReq(stop) received?, S2 switch opened(CP State C->B)?), HV DC current is enough low(depend on normal/fast/emergency) for safety?, ...
> 
> (2) As I know, PWM duty should(?) be retained(1.5s) for V2G communication(unless tcp socket closed by peer(EVCC)) in normal stop cases (according to DIN 70121 [V2G-DC-672]).
> 
> > [V2G-DC-672] After the SECC has sent a SessionStopRes message with ResponseCode equal to “OK”, it shall keep the CP oscillator switched on for V2G_SECC_CPOscillator_Retain_Time according to Table 77. The SECC shall then switch the CP oscillator off.
> 
> And, it might be more helpful if you attach the log(tcpdump,..) files of test cases(ex, good(?) case of 'Tesla Model 3/Y', bad case of 'Renault Zoe', .. not only for me, but also for others(@lho-stx )).

Hi, thanks for your answer

1) I do wait for "Stop Charging" request only if the EV ends the charging, not when the EVSE stops the charging (when for example the "Stop" button is pressed on the charger), for this case I send a notification to stop charging (0x75), then I send a stop charge request (0x6F).

When the Stop charging sequence is initiated from the EV or the EVSE, the current is ramped quickly down to 0A and the contactors opens only if the current is below 1A.

2) I also think that I should stop CP PWM at the very end of the session, not before stopping HPGP and SLAC.

I've to try this the next week, I will inform you of the results

pxidr, 2024-09-18:
So I tried with this sequence:
1.Send notification to stop charging (ModId 0x27 SubId 0x75)
2.Send stop charge (ModId 0x27 SubId 0x6F)
3.Wait that current is below 0.9A
4.Open all DC contactors on the charger side
5.Stop update parameters loop (ModId 0x27 SubId 0x63)
6.Stop listen to HPGP comms (ModId 0x27 SubId 0x70)
7.Stop SLAC (ModId 0x28 SubId 0x43)
8.Set PWM duty cycle to 100% (ModId 0x29 SubId 0x44)

Still, the Zoe won't unlock the connector quickly. Tried with an other EV (Megane E-tech), exactely the same issue.
Any ideas?

lho-stx, 2024-09-19:
Hey,

do you have log files (wireshark(!), UART, FreeV2G output) of the Zoe not release the lock? If yes please share them.

Thank you,
lho

pxidr, 2024-09-22:
Hello,

I don't have the Zoe anymore, so I can't show you the logs for now, sorry.
However I would like to know if my stop sequence is correct.

What I didn't check is if the car does a Contactor Welding test, that may explain why it won't release the plug.

lho-stx, 2024-09-24:
Hey @pxidr,

in general the sequence looks ok. Please double check the Welding test on your side.

Best regards,
lho