Sevenstax/FreeV2G #345: Question: There is a discrepancy with the documentation. What is  the problem?

Volodymyr233, 2024-09-27:
**Before you post**
- [ ] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [ ] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

I have a deal with EV ver. 01_00_04. After successful execution of command "Start Session" I received message: C0 27 C0 00 00 32 01 0F 52 85 C7 76 99 96 8E 26 44 45 2A 41 32 33 2A 45 34 35 42 2A 37 38 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 C1 (i.e. "Session Started"). In all other messages instead Req-ID presents FF. But here I see 00. Is it correct?
log-file: 
[output_2024-09-27_16-15-55.log](https://github.com/user-attachments/files/17164919/output_2024-09-27_16-15-55.log)

lho-stx, 2024-10-09:
Hey,

your are using an outdated version of the WHITE Beet firmware. Please consider updating and retry to run your charging session. Please also make sure to use the documentation matching with the firmware version used.

Best regards,
lho

Volodymyr233, 2024-10-10:
Got it, thanks.

closed 2024-10-25 by fri-stx