Sevenstax/FreeV2G #346: Question: Unable to Change Energy Transfer Mode to 4,5 for AC Charging

hamzamuneer95, 2024-10-18:
Hi, 

I am using  **WHITE-beet-EI firmware version: V02_01_00** and **WHITE-beet-PI firmware version: V01_00_07**. 

I am using the FreeV2G Stack, I configured both boards as per the instructions in the wiki and was successfull in establishing a connection between the two boards in DC Charging Mode.

The problem occures when I change the **Energy Transder Mode to 4** in the EVSE.py file. I have attached the screenshots of change I have made as well as the outputs of the both EVSE and PEV modules. 

I have tried using Energy Transfer Mode 4,5 and only 4 as well

[image]
[image]

lho-stx, 2024-10-29:
Hey @hamzamuneer95,

I am working on this, I am not quite sure what the problem is at the moment. I created a patch file that is enabling FreeV2G to continue the session, but now the EV terminates the session after ChargeParameterDiscoveryRes. I attach the patch file, so you can also apply it to your copy of FreeV2G.

[enable-ac-charging.PATCH](https://github.com/user-attachments/files/17559630/enable-ac-charging.PATCH)

Unfortunately I need more time to investigate the issue, but I think it is related to the FreeV2G configuration.

Best regards,
lho

fri-stx, 2024-10-30:
Hi @hamzamuneer95,

i continued working on the patch from @lho-stx. AC charging is now working with EV and EVSE. Please let me now if this also works for you so i can close the issue. Sorry for the inconvenience and thanks for the report.

Best regards,
fri

PS: Please use the "Bug report" template next time and provide log files as discribed in https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files . This will help us to analyze problems faster.

hamzamuneer95, 2024-10-31:
[image]

Hi @lho-stx @fri-stx i went through the patch file and found this logic confusing, please comment

hamzamuneer95, 2024-10-31:
Hi @fri-stx,

Thanks for sharing the update, the new patch file fixed most of the problems. However, there is still a change.  Line 557 in EV.py needs to be commented out as it is hardcoding the schedule and not using the one received from EVSE

fri-stx, 2024-11-01:
Hi @hamzamuneer95,

I will make this work and provide a patch file. All of these changes will be added to the master branch soon.

Best regards,
fri

fri-stx, 2024-11-01:
Hi @hamzamuneer95,

this is the patch file with the removed line.

[fixed-ac-charging_final.patch](https://github.com/user-attachments/files/17600464/fixed-ac-charging_final.patch)

Best regards,
fri