Sevenstax/FreeV2G #348: Question: Insurmountable process stop

Volodymyr233, 2024-10-23:
**Before you post**
- [+ ] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [+ ] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

I can't pair two devices: WhiteBeet EV and WhiteBeet EVSE. The process reaches the "Session Started" messages from both devices. Then the EVSE takes a few more steps:

Message: Payment Selected
Message: Autorization Status Requested -> Command: Set Autorization Status
Message: Energy Transfer Mode Selected
Message: Schedules Requested -> Command: Set Schedules
Message: DC Charge Parameters Changed -> Repeatable Command: EVSE Update DC Charging Parameters.

EV waits timeout and then breaks session.

I provide a couple of log files for each side. One file is received via the UART port of the WhiteBeet debug board. The second is an interception of the SPI exchange between the host and the debug board. I cleaned up the duplicate commands. What I noticed: EVSE accepted DC charging parameters from EV. But EV did not. Although Session ID, Device ID were accepted! That is, the physical connection works. I do not quite understand at what point to start repeating "Update DC Charging Parameters" command on the EVSE side. I do this when receiving a message from EV "DC Charge Parameters Changed". 

Best regards, Volodymyr.

Log-files for EV side:
[EV_2024-10-21_18-11-10.log](https://github.com/user-attachments/files/17491410/EV_2024-10-21_18-11-10.log)
[EV_21101810.log](https://github.com/user-attachments/files/17491411/EV_21101810.log)
Log-files for EVSE side:
[EVSE_2024-10-21_18-11-02.log](https://github.com/user-attachments/files/17491415/EVSE_2024-10-21_18-11-02.log)
[EVSE_21101810.log](https://github.com/user-attachments/files/17491416/EVSE_21101810.log)

lho-stx, 2024-10-29:
Hey @Volodymyr233,

due to the log files not beeing wireshark log files we need some more time for inspecting.

But I can tell you that need to send the "Update DC Charging Parameters" on EVSE side after the schedules are requested by the EV.

Please allow some more time for investigation.

Best regards,
lho

lho-stx, 2024-12-16:
Hey @Volodymyr233,

Are you using a custom host application? I can see an issue with the "Update DC Charging Paremeters" message. The framing payload is:

```
C0 27 63 1A 00 11 01 00 00 00 00 00 00 00 26 01 00 05 01 00 13 03 00 00 C1
```

Because `Max. voltage`,` Max. current` and `Max. power` are optional they have a boolean in front that tells if the parameter is present. This means your message should be:

```
C0 27 63 1A 00 11 01 00 00 00 00 00 00 00 01 26 01 00 01 05 01 00 01 13 03 00 00 C1
```

Please try to adapt your custom host application and let me know if you still have problems.

Best regards,
lho

Volodymyr233, 2024-12-17:
Thanks, I will try.