Sevenstax/FreeV2G #349: WHITE-BEET Module fails to start V2G service

dtomasi95, 2024-10-29:
Platform: PEV (WHITE-BEET-PI (EV) Module with Embedded ISO15118 Sw Stack )
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet
Host: Own Implementation

**Describe the bug**
Recently we discovered an issue associated with the white-beet module that we never encounter before, and we don’t have a clear view on how to resolve it.
The issue is correlated with the starting of the V2G service (Module-ID: 0x27):
as indicated in the documentation provided with the white-beet module and as done in the example of host controller implementation present in the Sevenstax repository: 
Upon receiving the SLAC process was successful message (Module-ID: 0x28 + Sub-ID:0x80) from the white-beet module, our device acting as the host controller sends in sequence the following messages:

1. **Set Mode to EV** (Module-ID: 0x27 + Sub-ID:0x40)
2. **Set Configuration** (Module-ID: 0x27 + Sub-ID:0xa0
3. **Set DC Charging Parameters** (Module-ID: 0x27 + Sub-ID:0xa2)
4. **Start Session** (Module-ID: 0x27 + Sub-ID:0xa9)

**NOTE:** Our device waits for a response from the white-beet module with the same Module-ID + Sub-ID before sending the next command, but if the any result code is negative, it still sends all the commands in the above block (our charging test is interrupted if any result code is negative).

If everything goes well each of the commands responds with the result code 0x00: Command accepted, and the charging session starts.
But sometimes the white-beet module responds with the following pattern:
1. **Set Mode to EV** (Module-ID: 0x27 + Sub-ID:0x40)  --> **0x01: Module is busy; try again later**
2. **Set Configuration** (Module-ID: 0x27 + Sub-ID:0xa0 -->  **0x11 Wrong state, not in the state to receive the command**
3. **Set DC Charging Parameters** (Module-ID: 0x27 + Sub-ID:0xa2) --> **0x11 Wrong state, not in the state to receive the command**
4. **Start Session** (Module-ID: 0x27 + Sub-ID:0xa9) --> **0x11 Wrong state, not in the state to receive the command**

**To Reproduce**
The issue is difficult to reproduce since it happens sporadically.
That makes debugging it quite difficult and time-consuming therefore we hope to leverage your insight on the product to quickly find the origin of the issue and deploy a solution.

**Expected behavior**
Each of the V2G service commands responds with the result code **0x00: Command accepted**, and the charging session starts.
as dose on all the charging sesion where the issue is not present. 

**Logs**
the dificulty of reproducing the issue prevented us from having a complete log set.
following there is the log of the comunication between our host controller and the white-beet module
[image]

**Additional context**
The issue has been reported after we deployed a new series of devices that are equipped with a white-beet module with firmware version: 01_00_07.
We can’t associate for sure the issue with the new version, since we were unable to reproduce the issue enough times to come to a certain conclusion.
Maybe there is something that we can change on our side to avoid the issue.
We can only say that as for now we never noticed the issue with the previous version that we used: v01_00_05.

Another thing that we noticed is that the V2G service seems to become stuck when this issue happens.
What I mean is that we could have an arbitrary number of charging sessions where everything is working fine, then this issue happens and after that it will keep happening in any charging session.
Even after changing CCS outlet or charger.
It disappears only after rebooting the white-beet module by disconnecting it from the power source.
After a white-beet module reboot it starts to work fine again until it happens next time.

Just to mention it, we did try to do a downgrade to see if the issue persisted on the previous version but we were unable to do so with the firmware updater tool provided by Sevenstax.
The tool seems to report a completed update but after checking the firmware version on the device with the appropriate command we still see as the reported version: v01_00_07.

lho-stx, 2024-10-29:
Hey @dtomasi95,

thank you for all the information although you do not have the log files at hand!

I need to look into this in detail, but please try to send a StopSession frame before adjusting the charging settings. This should resolve the 

```
0x01: Module is busy; try again later
```

error and all subsequent error messages. I suspect that the StopSession is somehow not send or seen at the correct moment, therefore your EV V2G service is still running.

You can try to reproduce the error by configuring your EV, starting the V2G service with StartSession and then try the configuration again. If you get the same error as above, then your V2G service is still running. You then should check your implementation if the StopSession is send each and every time before the configuration of the EV is changed.

Please try it and let me know your results.

Regarding the downgrade:  I do not recommend a downgrade at this point. But you can try an upgrade to the 1.1.0 firmware on one of your boards. It enables PnC, but is compatible to your solution you already have. 

Best regards,
lho

dtomasi95, 2025-01-10:
Hi @lho-stx,
Thanks for your reply and sorry for the wait, unfortunately in this last period I had bigger fires to put out.

Before the holidays I managed to spend some time in production and test your solution.

I edited my program to make sure the StopSession frame is sent when the power test reset so it is guaranteed every time and then waited to see if it happened again.
```
15:35:32.762091 [|decnet]
	0x0000:  c493 0047 cd45 f8dc 7abb 2ed8 6003 0004
	0x0010:  0008 c027 ae01 0000 a6c1 0000 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:32.763423 [|decnet]
	0x0000:  f8dc 7abb 2ed8 c493 0047 cd45 6003 0004
	0x0010:  0009 c027 ae01 0001 00a5 c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
``` 

You were right your solution solved the error: 
```
0x01: Module is busy; try again later

15:35:42.654564 [|decnet]
	0x0000:  c493 0047 cd45 f8dc 7abb 2ed8 6003 0004
	0x0010:  0009 c027 402a 0001 00ea c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:42.655988 [|decnet]
	0x0000:  f8dc 7abb 2ed8 c493 0047 cd45 6003 0004
	0x0010:  0009 c027 402a 0001 00ea c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
``` 
But unfortunately, the subsequent errors persisted:
```
0x11 Wrong state, not in the state to receive the command

15:35:42.660550 [|decnet]
	0x0000:  c493 0047 cd45 f8dc 7abb 2ed8 6003 0004
	0x0010:  0017 c027 a02b 000f c493 0047 cd45 0100
	0x0020:  0100 0101 1388 0128 c100 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:42.662041 [|decnet]
	0x0000:  f8dc 7abb 2ed8 c493 0047 cd45 6003 0004
	0x0010:  0009 c027 a02b 0001 1178 c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:42.678051 [|decnet]
	0x0000:  c493 0047 cd45 f8dc 7abb 2ed8 6003 0004
	0x0010:  002b c027 a22c 0023 0000 0000 0000 0000
	0x0020:  0003 e800 00fa 000f a002 0000 0190 0000
	0x0030:  0500 6464 012c 0001 0000 0638 c1
15:35:42.679052 [|decnet]
	0x0000:  f8dc 7abb 2ed8 c493 0047 cd45 6003 0004
	0x0010:  0009 c027 a22c 0001 1175 c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:42.685125 [|decnet]
	0x0000:  c493 0047 cd45 f8dc 7abb 2ed8 6003 0004
	0x0010:  0008 c027 a92d 0000 7fc1 0000 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
15:35:42.685730 [|decnet]
	0x0000:  f8dc 7abb 2ed8 c493 0047 cd45 6003 0004
	0x0010:  0009 c027 a92d 0001 116d c100 0000 0000
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0030:  0000 0000 0000 0000 0000 0000
``` 

and they persisted across multiple sessions until a hard reboot of the WHITE-BEET
I attached the full session dump if you want to give it a look.
[frameDump.txt](https://github.com/user-attachments/files/18376993/frameDump.txt)
[whitebeetDump.zip](https://github.com/user-attachments/files/18377226/whitebeetDump.zip)

Thanks for your help.
Best regards.